NixOS/nixos-hardware #1318: How to build (sd-)image with Btrfs root filesystem?

onny, 2025-01-16:
Hey,
would like to build an image with Btrfs as root filesystem for the VisionFive2 since I'm using NixOS on a NVME drive and would like to use Btrfs snapshot and send/receive features on the system.

My current approach will be compiling U-Boot with Btrfs FS support and converting the existing Ext4 rootfs to Btrfs using btrfs-convert.

Regards
Jonas

Mic92, 2025-01-17:
Maybe disko-install assuming uboot is compatible with nixos-install?

RadxaYuntian, 2025-01-20:
I think it is better to just use EFI boot from U-Boot if you want alternative rootfs filesystem. I'm currently experimenting this and got bcachefs working on ROCK 4C Plus.

Mic92, 2025-01-20:
There is edk2 for UEFI: https://github.com/starfive-tech/edk2 but can't tell how easy and reliable it is to flash to the visionfive2. Given that these boards already have an EMMC, I wished they would just ship it with preflashed emmc/edk2.