NixOS/nixos-hardware #1330: Attribute `config.boot.kernelPackages` missing in `framework/kmod.nix`

wiggleforlife, 2025-01-20:
Hi, I updated my flake and rebuilt my config on my FW16, and now I get this:
```error:
       … while calling the 'head' builtin
         at /nix/store/l9nb64iii15y0nr37qrs1cfm6rlpg6gh-source/lib/attrsets.nix:1574:11:
         1573|         || pred here (elemAt values 1) (head values) then
         1574|           head values
             |           ^
         1575|         else

       … while evaluating the attribute 'value'
         at /nix/store/l9nb64iii15y0nr37qrs1cfm6rlpg6gh-source/lib/modules.nix:846:9:
          845|     in warnDeprecation opt //
          846|       { value = addErrorContext "while evaluating the option `${showOption loc}':" value;
             |         ^
          847|         inherit (res.defsFinal') highestPrio;

       … while evaluating the option `system.build.toplevel':

       … while evaluating definitions from `/nix/store/l9nb64iii15y0nr37qrs1cfm6rlpg6gh-source/nixos/modules/system/activation/top-level.nix':

       … while evaluating the option `system.systemBuilderArgs':

       … while evaluating definitions from `/nix/store/l9nb64iii15y0nr37qrs1cfm6rlpg6gh-source/nixos/modules/system/activation/activatable-system.nix':

       … while evaluating the option `system.activationScripts.etc.text':

       … while evaluating definitions from `/nix/store/l9nb64iii15y0nr37qrs1cfm6rlpg6gh-source/nixos/modules/system/etc/etc-activation.nix':

       … while evaluating definitions from `/nix/store/l9nb64iii15y0nr37qrs1cfm6rlpg6gh-source/nixos/modules/system/etc/etc.nix':

       … while evaluating the option `environment.etc.dbus-1.source':

       … while evaluating the default value of option `hardware.framework.enableKmod`

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: attribute 'kernelPackages' missing
       at /nix/store/1yy1mqy4jj3xy298p6svvqr6s8gqdy3l-source/framework/kmod.nix:3:50:
            2| let
            3|   kernel_version_compatible = lib.versionAtLeast config.boot.kernelPackages.kernel.version "6.10";
             |                                                  ^
            4| in {
```

arunoruto, 2025-01-22:
I get the following error with `hardware.framework.enableKmod = true`:
```log
framework-laptop-kmod> building '/nix/store/5p553m6268qlkgbiby397089vh4gqz22-framework-laptop-kmod-0-unstable-2024-09-15.drv'
framework-laptop-kmod> Running phase: unpackPhase
framework-laptop-kmod> unpacking source archive /nix/store/i2vrg24drigcm8ixhqs1qwbvnmd5iwgl-source
framework-laptop-kmod> source root is source
framework-laptop-kmod> Running phase: patchPhase
framework-laptop-kmod> Running phase: updateAutotoolsGnuConfigScriptsPhase
framework-laptop-kmod> Running phase: configurePhase
framework-laptop-kmod> no configure script, doing nothing
framework-laptop-kmod> Running phase: buildPhase
framework-laptop-kmod> build flags: SHELL=/nix/store/gwgqdl0242ymlikq9s9s62gkp5cvyal3-bash-5.2p37/bin/bash O=\$\(buildRoot\) CC=/nix/store/888bkaqdpfpx72dd8bdc69qsqlgbhcvf-gcc-wrapper-13.3.0/bin/cc HOSTCC=/nix/store/888bkaqdpfpx72dd8bdc69qsqlgbhcvf-gcc-wrapper-13.3.0/bin/cc HOSTLD=/nix/store/qlzvmgr8w9prdlyys7irqf86p7bndf5b-binutils-wrapper-2.43.1/bin/ld ARCH=x86_64 KDIR=/nix/store/yams2a5wc4qg2i815affb5yh1pbwgf3k-linux-6.13-dev/lib/modules/6.13.0/build
framework-laptop-kmod> make -C /nix/store/yams2a5wc4qg2i815affb5yh1pbwgf3k-linux-6.13-dev/lib/modules/6.13.0/build M=$PWD modules
framework-laptop-kmod> make[1]: Entering directory '/nix/store/yams2a5wc4qg2i815affb5yh1pbwgf3k-linux-6.13-dev/lib/modules/6.13.0/build'
framework-laptop-kmod> mkdir: cannot create directory '.tmp_63': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_66': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_69': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_72': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_75': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_78': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_81': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_84': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_87': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_90': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_93': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_96': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_99': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_102': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_105': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_108': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_111': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_114': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_117': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_120': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_123': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_126': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_129': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_132': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_135': Permission denied
framework-laptop-kmod> mkdir: cannot create directory '.tmp_138': Permission denied
framework-laptop-kmod> /nix/store/yams2a5wc4qg2i815affb5yh1pbwgf3k-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:777: /nix/store/yams2a5wc4qg2i815affb5yh1pbwgf3k-linux-6.13-dev/lib/modules/6.13.0/source/include/config/auto.conf: No such file or directory
framework-laptop-kmod> make[1]: *** [/nix/store/yams2a5wc4qg2i815affb5yh1pbwgf3k-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:251: __sub-make] Error 2
framework-laptop-kmod> make[1]: Leaving directory '/nix/store/yams2a5wc4qg2i815affb5yh1pbwgf3k-linux-6.13-dev/lib/modules/6.13.0/build'
framework-laptop-kmod> make: *** [Makefile:12: modules] Error 2
```

Turning `enableKmod` off, makes it work again...

emillassen, 2025-01-22:
I'm getting the same error using the flake  `nixos-hardware.nixosModules.framework-13-7040-amd` 

```
Running phase: unpackPhase
@nix { "action": "setPhase", "phase": "unpackPhase" }
unpacking source archive /nix/store/i2vrg24drigcm8ixhqs1qwbvnmd5iwgl-source
source root is source
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: configurePhase
@nix { "action": "setPhase", "phase": "configurePhase" }
no configure script, doing nothing
Running phase: buildPhase
@nix { "action": "setPhase", "phase": "buildPhase" }
build flags: SHELL=/nix/store/4fvc5fm8bszmkydng1ivrvr5cbvr1g60-bash-5.2p37/bin/bash O=\$\(buildRoot\) ARCH=x86_64 CROSS_COMPILE= KDIR=/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build
make -C /nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build M=$PWD modules
make[1]: Entering directory '/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build'
mkdir: cannot create directory '.tmp_63': Permission denied
mkdir: cannot create directory '.tmp_66': Permission denied
mkdir: cannot create directory '.tmp_69': Permission denied
mkdir: cannot create directory '.tmp_72': Permission denied
mkdir: cannot create directory '.tmp_75': Permission denied
mkdir: cannot create directory '.tmp_78': Permission denied
mkdir: cannot create directory '.tmp_81': Permission denied
mkdir: cannot create directory '.tmp_84': Permission denied
mkdir: cannot create directory '.tmp_87': Permission denied
mkdir: cannot create directory '.tmp_90': Permission denied
mkdir: cannot create directory '.tmp_93': Permission denied
mkdir: cannot create directory '.tmp_96': Permission denied
mkdir: cannot create directory '.tmp_99': Permission denied
mkdir: cannot create directory '.tmp_102': Permission denied
mkdir: cannot create directory '.tmp_105': Permission denied
mkdir: cannot create directory '.tmp_108': Permission denied
mkdir: cannot create directory '.tmp_111': Permission denied
mkdir: cannot create directory '.tmp_114': Permission denied
mkdir: cannot create directory '.tmp_117': Permission denied
mkdir: cannot create directory '.tmp_120': Permission denied
mkdir: cannot create directory '.tmp_123': Permission denied
mkdir: cannot create directory '.tmp_126': Permission denied
mkdir: cannot create directory '.tmp_129': Permission denied
mkdir: cannot create directory '.tmp_132': Permission denied
mkdir: cannot create directory '.tmp_135': Permission denied
mkdir: cannot create directory '.tmp_138': Permission denied
/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:777: /nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/include/config/auto.conf: No such file o>
make[1]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build'
make: *** [Makefile:12: modules] Error 2
```

fzakaria, 2025-01-22:
To circumvent for now
```
  hardware = {
    # disable framework kernel module
    # https://github.com/NixOS/nixos-hardware/issues/1330
    framework.enableKmod = false;
  };
```

marienz, 2025-01-25:
The build failure on 6.13 in the last few comments looks like https://github.com/NixOS/nixpkgs/issues/376324 (triggered by the Linux 6.13 upgrade in nixpkgs, separate issue from the evaluation failure involving `config.boot.kernelPackages` this issue was opened for).

wiggleforlife, 2025-01-25:
Ah, I'm on 6.12. I can build kmod still by pinning to the commit before https://github.com/NixOS/nixos-hardware/commit/ce98bebf0f98140cb1d4b6a3edee4f1eaade9261 like @gfauredev suggested. Are people on 6.13 building with the latest version or a pinned version?

lvstb, 2025-01-28:
> To circumvent for now
> 
> ```
>   hardware = {
>     # disable framework kernel module
>     # https://github.com/NixOS/nixos-hardware/issues/1330
>     framework.enableKmod = false;
>   };
> ```

For me this didnt solve it.

nyabinary, 2025-01-30:
It should default to off for kernels 6.12 and above except for 12/13th gen Intel CPUs, since the kmod isn't needed for the other platforms (functionality got upstreamed).

Undercoverer, 2025-02-06:
Creating a channel for a working commit has worked for me in the meantime. 
```
sudo nix-channel --add https://github.com/NixOS/nixoshardware/archive/e7ac747157a3301034b0caea9eb45c7b071e52fd.tar.gz nixos-hardware-pinned
```

xeniarose, 2025-03-25:
It seems like there's 2 different bugs: one upstream bug which affected all out-of-tree kernel modules on 6.13, which has since been fixed; and the original bug in the OP where `boot.kernelPackages` appears to be missing.

I'm trying to reproduce that issue (2 months later) and I cannot reproduce it. I don't see any reason that `boot.kernelPackages` would be completely missing on a Framework 16 NixOS build. Is there an example NixOS config that triggers the issue?