NixOS/nixos-hardware #1352: GPD Win Max 2 (2023/2024): bmi-260 doesn't compile for linux 6.13

toast003, 2025-01-31:
Build logs:
```
Running phase: unpackPhase
@nix { "action": "setPhase", "phase": "unpackPhase" }
unpacking source archive /nix/store/idi8im8byv5wlkfjrcilsh4ncnqgd899-source
source root is source
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "patchPhase" }
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScriptsPhase" }
Running phase: configurePhase
@nix { "action": "setPhase", "phase": "configurePhase" }
no configure script, doing nothing
Running phase: buildPhase
@nix { "action": "setPhase", "phase": "buildPhase" }
build flags: SHELL=/nix/store/4fvc5fm8bszmkydng1ivrvr5cbvr1g60-bash-5.2p37/bin/bash KERNEL_SRC=/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/l>
make -C /nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build M=/build/source modules
make[1]: Entering directory '/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build'
make[2]: Entering directory '/build/source'
  CC [M]  bmi260_core.o
/build/cc96wnTx.s: Assembler messages:
/build/cc96wnTx.s:7: Error: junk at end of line, first unrecognized character is `I'
/build/cc96wnTx.s:8: Error: junk at end of line, first unrecognized character is `I'
/build/cc96wnTx.s:9: Error: junk at end of line, first unrecognized character is `I'
make[4]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/scripts/Makefile.build:194: bmi260_core.o] Error 1
make[3]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:1989: .] Error 2
make[2]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/build/source'
make[1]: *** [/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/source/Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/nix/store/6b9j17qfxdvyh9kmi9fjgh6zg0ix8if5-linux-6.13-dev/lib/modules/6.13.0/build'
make: *** [Makefile:15: modules] Error 2
```

toast003, 2025-01-31:
@Cryolitia

Cryolitia, 2025-02-05:
Report it upstream: https://github.com/justinweiss/bmi260/issues/1

closed 2025-03-12 by mergify[bot]