NixOS/nixos-hardware #1388: Lenovo Legion 15ach6h : amdgpuBusId seems incorrect

Malix-Labs, 2025-02-23:
On my machine (a Lenovo Legion 15ach6h), the amdgpuBusId seems to be `PCI:5:0:0` instead of `PCI:6:0:0`

## Subject

https://github.com/NixOS/nixos-hardware/blob/18e9f9753e9ae261bcc7d3abe15745686991fd30/lenovo/legion/15ach6h/hybrid/default.nix#L31C9-L31C20

## Log

```sh
[nix-shell:~]$ fastfetch
          ▗▄▄▄       ▗▄▄▄▄    ▄▄▄▖             malix@nixos
          ▜███▙       ▜███▙  ▟███▛             -----------
           ▜███▙       ▜███▙▟███▛              OS: NixOS 24.11 (Vicuna) x86_64
            ▜███▙       ▜██████▛               Host: 82JU (LEGION 5 15ACH6H)
     ▟█████████████████▙ ▜████▛     ▟▙         Kernel: Linux 6.13.2
    ▟███████████████████▙ ▜███▙    ▟██▙        Uptime: 8 mins
           ▄▄▄▄▖           ▜███▙  ▟███▛        Packages: 1094 (nix-system)
          ▟███▛             ▜██▛ ▟███▛         Shell: bash 5.2.37
         ▟███▛               ▜▛ ▟███▛          Display (AUO5895): 1920x1080 @ 165 Hz in 15" [Built-in] *
▟███████████▛                  ▟██████████▙    Display (ASUSTek COMPUTER INC 27"): 2560x1440 @ 120 Hz in 27" [External]
▜██████████▛                  ▟███████████▛    DE: GNOME 47.2
      ▟███▛ ▟▙               ▟███▛             WM: Mutter (Wayland)
     ▟███▛ ▟██▙             ▟███▛              WM Theme: Adwaita
    ▟███▛  ▜███▙           ▝▀▀▀▀               Theme: Adwaita [GTK2/3/4]
    ▜██▛    ▜███▙ ▜██████████████████▛         Icons: Adwaita [GTK2/3/4]
     ▜▛     ▟████▙ ▜████████████████▛          Font: Cantarell (11pt) [GTK2/3/4]
           ▟██████▙       ▜███▙                Cursor: Adwaita (24px)
          ▟███▛▜███▙       ▜███▙               Terminal: GNOME Console 47.1
         ▟███▛  ▜███▙       ▜███▙              Terminal Font: Source Code Pro (10pt)
         ▝▀▀▀    ▀▀▀▀▘       ▀▀▀▘              CPU: AMD Ryzen 7 5800H (16) @ 4.46 GHz
                                               GPU 1: NVIDIA GeForce RTX 3060 Mobile / Max-Q [Discrete]
                                               GPU 2: AMD Radeon Vega Series / Radeon Vega Mobile Series [Integrated]
                                               Memory: 2.07 GiB / 13.51 GiB (15%)
                                               Swap: 0 B / 14.86 GiB (0%)
                                               Disk (/): 12.97 GiB / 922.70 GiB (1%) - ext4
                                               Local IP (eno1): 192.168.1.50/24
                                               Battery (L20D4PC1): 100% [AC Connected]
                                               Locale: en_US.UTF-8

                                                                       
                                                                       

[nix-shell:~]$ lspci
00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:02.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:08.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 51)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 7
01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GA106 High Definition Audio Controller (rev a1)
02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 15)
04:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852AE 802.11ax PCIe Wireless Network Adapter
05:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c5)
05:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
05:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne USB 3.1
05:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne USB 3.1
05:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] ACP/ACP3X/ACP6x Audio Coprocessor (rev 01)
05:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h/19h/1ah HD Audio Controller
06:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
06:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
```

Malix-Labs, 2025-02-23:
@Lyndeno @yuannan

Mic92, 2025-02-24:
Sometimes Lenovo has different versions of the same model depending on the location you buy it.
We noticed it with X1 laptops bought in Europe vs UAE.

Malix-Labs, 2025-02-24:
Would there be a way to detect the PCI ports dynamically?

yuannan, 2025-03-06:
Hello, sorry for the late reply. I've been pretty busy with work and some personal stuff.

On my machine `6:00.0` is the right ID. Here is the full ID if you want to look at it.

```
> doas lspci

00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne IOMMU
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:01.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe GPP Bridge
00:01.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:02.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:02.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:02.4 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne PCIe GPP Bridge
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Renoir PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:08.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Renoir Internal PCIe GPP Bridge to Bus
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 51)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Cezanne Data Fabric; Function 7
01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GA106 High Definition Audio Controller (rev a1)
02:00.0 Non-Volatile memory controller: Phison Electronics Corporation E12 NVMe Controller (rev 01)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet Controller (rev 15)
04:00.0 Network controller: Intel Corporation Wi-Fi 6 AX200 (rev 1a)
05:00.0 Non-Volatile memory controller: Sandisk Corp WD Black SN850X NVMe SSD (rev 01)
06:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Cezanne [Radeon Vega Series / Radeon Vega Mobile Series] (rev c5)
06:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 10h-1fh) Platform Security Processor
06:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne USB 3.1
06:00.4 USB controller: Advanced Micro Devices, Inc. [AMD] Renoir/Cezanne USB 3.1
06:00.5 Multimedia controller: Advanced Micro Devices, Inc. [AMD] ACP/ACP3X/ACP6x Audio Coprocessor (rev 01)
06:00.6 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h/19h/1ah HD Audio Controller
07:00.0 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)
07:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 81)                                                                                 
```

I'm pretty busy with work for the next few weeks, but I cannot see why we couldn't do this dynamically, not just for this, but the entire repo via a "compile time" script. I think if we were to do this, a simple pattern match should work. The prefix seems to be always `VGA`, followed by the name of the GPU. If this is different on different kernels and systems is not information that I know currently. If you create a new ticket/issue for tracking `lspci` outputs, with the goal to generalise this ID matching for all systems supported in this repo, I would be happy to contribute to a PR once I've got the time to do so.

This can be merged into the higher level nvidia and amd files, with lower level dedicated nix files such as this for specific machines as overwrites. This should be fairly trivial with default values that nix supports.

Personally, I've just switched to just using the Nvidia GPU only as I've had loads of random crashes and during gaming and general usage trying to use both GPUs. The only real advantage that I saw from using both is potentially lower power usage, as well as VRAM usage. But overall, I've found using just the Nvidia GPU is the best for stability and performance. If that is something that you could do, it's the best way to use this laptop for me. I've found that I'm rarely requiring the lower power usage of shutting down the Nvidia GPU. It's a gaming laptop so I'm pretty much always plugged in.

Mic92, 2025-03-07:
If we can do this more dynamically, this should be preferred over our current approach.

yuannan, 2025-03-08:
Also, as per the README.md (https://github.com/NixOS/nixos-hardware/tree/master/lenovo/legion/15ach6h#using-multiple-drives-with-this-configuration) this is due to having more than 1 drive.

Should we change the default ID to assume users only have 1 drive instead of 2? I don't really know the statistics, but I personally dual boot with Windows, and even if I didn't, I'd probably have 2 drives anyway.

yuannan, 2025-03-10:
There are some docs here that could facilitate this (https://nix.dev/manual/nix/2.26/language/import-from-derivation). 

@Mic92 You seem to be one of the largest maintainers here, what are your thoughts on doing something like this? In my honest opinion this could cause more problems then it solves. This is really a vendor (nvidia) issue. I don't think it's user friendly to inspect the IDs manually and set it. In an ideal world the nvidia driver would be able to see that it has the right GPUs and set all of this info at the driver level instead.

However, I don't think the current approach is very user friendly.

The right approach to be to do this at runtime instead of trying to set all of this statically, that includes detecting the PCI ports using a script as this will at the very minium require the user to rebuild their system to get the right config.

imo, this is either a can't fix/won't fix as it goes against the very philosophy of nix.

Mic92, 2025-03-10:
Mhm. If you try to use import-from-derivation here, I don't think this is feasible for two reasons:

1. I don't think nix currently allows nix builds to access hardware device nodes in it's sandbox
2. If it was possible to access this information, It would break if someone uses remote building on a different machine.

The configuration that uses this information seems to be passed directly to the xorg server via commandline flag. I haven't seen a way to add dynamic configuration to that file.
So maybe it's not so trivial to make this information more dynamic.
So should we just stick to the more common configuration and than document that people might have to change this value?

Lyndeno, 2025-04-07:
I am in favour of user friendliness when it is possible. In this case I do not think it is.

Options I can think of:
1. Leave the default common value, document the possible deviation
2. Build-time check that goes against the sandbox
3. Do not set the value, require the user set a value or throw an error. Ship a simple script that the user can run manually that tells them what they need to put in the config.

I think option 3 is best. With option 1 some people may not read the docs, get an error when their config does not work. Then open an issue here.

Option 2 breaks the sandbox if it works at all.

Option 3 will tell the user what they need to do if they do not supply a value.

Malix-Labs, 2025-04-07:
If the sandbox doesn't allow a build-time check, does it have a system of declarative specific permission (à la flatpak)?

If not, are there any open issue / proposal about this?
Else, don't you think such permission system should be created?

Malix-Labs, 2025-05-31:
Hello again folks;
Can I be of any help with the runtime port check?

yuannan, 2025-06-01:
@Malix-Labs I don't think it's as easy as it first appears. As pointed out in previous posts having these configs run at "compile" time does more harm than good. We'll at the very least have to fully rebuild this pkg every time the config changes. As Nix caches it's prebuilt pkgs, I don't think there is an easy way to do this.

I've got myself an AMD laptop now, so yeah...

With the current hardware market, I doubt I'll ever be getting an Nvidia GPU ever again.

Good luck 😉

Malix-Labs, 2025-06-01:
On another note: `lenovo-legion-15ach6h-hybrid` doesn't exist and there is no documentation about how to explicitely use the hybrid or the nvidia module

 https://github.com/NixOS/nixos-hardware/tree/master/lenovo/legion/15ach6h

```
[malix@nixos:/etc/nixos]$ sudo nixos-rebuild switch --flake .#nixos
error:
       … while calling the 'seq' builtin
         at /nix/store/01635adignbizr4nyz5lpawkrxn4n4g8-source/lib/modules.nix:359:18:
          358|         options = checked options;
          359|         config = checked (removeAttrs config [ "_module" ]);
             |                  ^
          360|         _module = checked (config._module);

       … while evaluating a branch condition
         at /nix/store/01635adignbizr4nyz5lpawkrxn4n4g8-source/lib/modules.nix:295:9:
          294|       checkUnmatched =
          295|         if config._module.check && config._module.freeformType == null && merged.unmatchedDefns != [ ] then
             |         ^
          296|           let

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: attribute 'lenovo-legion-15ach6h-hybrid' missing
       at /nix/store/bhg6a37fksa2ry4kjbg82gii7cllhkjk-source/flake.nix:18:9:
           17|         # https://github.com/NixOS/nixos-hardware/tree/master/lenovo/legion/15ach6h
           18|         nixos-hardware.nixosModules.lenovo-legion-15ach6h-hybrid
             |         ^
           19|         ./nixos-hardware-override.nix
       Did you mean lenovo-legion-16ach6h-hybrid?

[malix@nixos:/etc/nixos]$ cat flake.nix
# /etc/nixos/flake.nix
{
  description = "NixOS configuration with flakes";

  inputs = {
    # Pin nixpkgs to the stable 25.05 release branch for better compatibility.
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-25.05";
    nixos-hardware.url = "github:NixOS/nixos-hardware/master";
  };

  outputs = { self, nixpkgs, nixos-hardware }: {
    nixosConfigurations.nixos = nixpkgs.lib.nixosSystem {
      system = "x86_64-linux";
      modules = [
        ./configuration.nix
        ./hardware-configuration.nix
        # https://github.com/NixOS/nixos-hardware/tree/master/lenovo/legion/15ach6h
        nixos-hardware.nixosModules.lenovo-legion-15ach6h-hybrid
        ./nixos-hardware-override.nix
      ];
    };
  };
}

[malix@nixos:/etc/nixos]$
```

yuannan, 2025-06-04:
> Should this mention be also removed?

No, this is accurate information. I think the purpose of this repo is to provide a "one size fits all(most)" config so users can use.

If you can write Nix I would recommend just using this repo as a base and then implementing it in your own configs straight up. Doing it raw is cheaper, faster, and more resistant to disruptions as you are no longer dependent on this repo.

In hindsight I think this issue is pretty fruitless as the needs of a few, be it me or you, will never be merged over the general public that needs packages to be built every time and uploaded into the binary cache.

imo, even if we were able to find a solution for this issues, I'm very doubtful it will be merged.

Malix-Labs, 2025-06-04:
> If you can write Nix I would recommend just using this repo as a base and then implementing it in your own configs straight up. Doing it raw is cheaper, faster, and more resistant to disruptions as you are no longer dependent on this repo.

I like the general concept about sharing configuration with others, thus i'm fine and even i'm happy with using this repo and trying to enhance it in case i become better at nix

I would be glad to eventually try to modularize `.ddg`, `.hybrid`, and `.nvidia` and then upstream it if that would be possible

> this issue is pretty fruitless as the needs of a few, be it me or you, will never be merged over the general public that needs packages to be built every time and uploaded into the binary cache.

I'm guessing you are talking about this current issue (as in, trying to resolve the problem about having two different pcie ids for the same model)

In that case, the two other options from @Lyndeno seems correct to me:

> Options I can think of:
>
> 1. Leave the default common value, document the possible deviation
> 2. Build-time check that goes against the sandbox
> 3. Do not set the value, require the user set a value or [...]

I think option 3 is the best

I was personally caught off-guard with the value being set by default on something other than `PCI:5:0:0`; In my experience it's the most common value i've encountered so I never had to change the default; and wasn't told I need to change it

`PCI:6:0:0` indeed seems to be when multiple drives are used

What do you all guys think?

4JX, 2025-06-13:
> It seems like you had a great idea indeed

I do want to work on it, but I just felt like there's enough variation within each config (be it common imports for specific generations or device quirks for example) that a common config would mostly solve PCI ID issues and not much more.

So I'm not sure it would cross the line of "useful abstraction".

Malix-Labs, 2025-09-16:
> So I'm not sure it would cross the line of "useful abstraction".

Maybe also using it for #1613 would justify it?

closed 2026-10-06 by mweinelt

Malix-Labs, 2026-11-14:
Can it be reopened?