NixOS/nixos-hardware #1433: Feature Request: Framework 13 Automatic fingerprint sensor switching

JustinLex, 2025-03-31:
Hi, I use the following snippet to enable and disable my fingerprint sensor when docking my Framework 13. Could this be added as an option to the Framework 13 module?

```nix
  services.acpid = {
    enable = true;
    lidEventCommands =
      let goodixVendor = "27c6";
      in ''
        grep -q closed /proc/acpi/button/lid/LID0/state
        if [ $? = 0 ]; then
          ${pkgs.fd}/bin/fd "-" /sys/bus/usb/devices --exec /bin/sh -c '${pkgs.gnugrep}/bin/grep -qs ${goodixVendor} {}/idVendor && ${pkgs.coreutils}/bin/echo 0 > {}/authorized'
        else
          ${pkgs.fd}/bin/fd "-" /sys/bus/usb/devices --exec /bin/sh -c '${pkgs.gnugrep}/bin/grep -qs ${goodixVendor} {}/idVendor && ${pkgs.coreutils}/bin/echo 1 > {}/authorized'
        fi
      '';
  };
```

Mic92, 2025-03-31:
Seems a bit opinionated to have this as a default? Some people prefer to even use the fingerprint when it's docked.

JustinLex, 2025-04-07:
Right, I understand that many users use the laptop docked with the lid open.

However, It's not possible to use the fingerprint reader when the lid is closed. This script does not affect users who use the laptop with the lid open.

The reason why I think this script is necessary is because with the default configuration, if a user has fingerprints set in GNOME, then sudo is not usable with the lid closed. Sudo will ask for a fingerprint, and the user will either need to open the lid to use the fprint reader, or wait for sudo to time out so that it asks for the password.

I thought that was pretty frustrating out of the box behavior, so I thought it would be good to add lid detection to the fingerprint reader. It is non-trivial to find the correct USB device to disable, so it would be good to share this script via nixos-hardware.

mdonoughe, 2025-07-06:
It's not obvious, but you can use sudo with the lid closed. I think it's if you press ^C at the fingerprint prompt.