NixOS/nixos-hardware #1439: Why does nixos-hardware support Apple Silicon if NixOS can’t run on macOS? Is this intended for nix-darwin?

phucleeuwu, 2025-04-06:
Hi, I’m a bit confused about the Apple Silicon modules in this repo.

As far as I know, NixOS doesn’t officially run on Apple Silicon devices yet

So I’m wondering:
- Are the Apple Silicon modules here intended for people running NixOS on Asahi Linux?
- Or are these modules more for nix-darwin users?
- Is there an official stance on NixOS support for Apple Silicon within nixos-hardware?

Would love clarification on the purpose or expected usage of these modules!

Thanks in advance.

Mic92, 2025-04-07:
What apple silicon modules are you referring to? In general it should be fine to have support for it in nixos-hardware.

phucleeuwu, 2025-04-07:
> What apple silicon modules are you referring to? In general it should be fine to have support for it in nixos-hardware.

mac silicon can't run nixos or other linux distro, mine is mac m4 mini

Mic92, 2025-04-24:
It works for some older devices fine: https://github.com/tpwrules/nixos-apple-silicon/tree/main