NixOS/nixos-hardware #1448: Orange Pi rv2 with X1 cpu

bartman, 2025-04-18:
Is there a known path to build an image for the Orange Pi RV2?

* [The official site](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-RV2.html)
* [User manual PDF](https://drive.google.com/drive/folders/1EAS0zgeR0cbZlLHeph5I_Y43rhKoS5af)

Booting from the provided Ubuntu 24.04 I can see...

```
Linux orangepirv2 6.6.63-ky #1.0.0 SMP PREEMPT Wed Mar 12 09:04:00 CST 2025 riscv64 riscv64 riscv64 GNU/Linux
```
(not sure yet if this "ky" CPU support is upstreamed)

<details><summary>lscpu</summary>

```
Architecture:          riscv64
  Byte Order:          Little Endian
CPU(s):                8
  On-line CPU(s) list: 0-7
Model name:            Ky(R) X1
  Thread(s) per core:  1
  Core(s) per socket:  8
  Socket(s):           1
  CPU(s) scaling MHz:  100%
  CPU max MHz:         1600.0000
  CPU min MHz:         614.4000
Caches (sum of all):
  L1d:                 256 KiB (8 instances)
  L1i:                 256 KiB (8 instances)
  L2:                  1 MiB (2 instances)
```
</details>

<details><summary>lspci</summary>

```
0001:00:00.0 PCI bridge: Device 201f:0001 (rev 01)
0002:00:00.0 PCI bridge: Device 201f:0001 (rev 01)
```
</details>

<details><summary>lsusb</summary>

```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 003 Device 003: ID 0781:55b1 SanDisk Corp. SanDisk 3.2 Gen1
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
</details>

<details><summary>lsmod</summary>

```
Module                  Size  Used by
hci_uart              458752  1
btrtl                 126976  1 hci_uart
btbcm                  73728  1 hci_uart
binfmt_misc            98304  1
sch_fq_codel          135168  4
bcmdhd               3321856  0
dhd_static_buf         32768  1 bcmdhd
ip_tables             122880  0
autofs4               315392  2
btrfs               14213120  0
blake2b_generic       131072  0
motorcomm             151552  2
```
</details>

Mic92, 2025-04-24:
Not quite the same but maybe it helps for inspiration: https://github.com/NixOS/nixos-hardware/pull/1409

vlinkz, 2025-04-24:
Doesn't seem like there's upstream support yet, would need to use custom kernel from https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-6.6-ky

misuzu, 2025-05-11:
I've hacked together something that boots:
https://gitlab.com/misuzu/hydra-riscv64/-/commit/f0074904a768816ed826dc59644736b4a6729e1c
https://hydra.ztier.in/build/10852

alekseysidorov, 2025-07-17:
I fixed the problems with Wifi in my repository, and add flake.nix.

https://github.com/alekseysidorov/nixos-orangepi-rv2

Now I am in the process of checking the GPU.

Later I can upstream my work to this repository.