NixOS/nixos-hardware #1465: Consider not enabling TLP by default

jolars, 2025-04-30:
Consider dropping TLP by default for laptops. Modern laptops usually have firmware that enables power profiles, so it often no longer relevant. Other times the distro has its own way of handling power profiles, like gnome, which is already handled here:

https://github.com/NixOS/nixos-hardware/blob/72081c9fbbef63765ae82bff9727ea79cc86bd5b/common/pc/laptop/default.nix#L9-L10

I am not entirely sure what the scope of nixos-hardware is, but to me TLP seems like an opinionated choice that is not objectively good for every laptop. Searching around the issues on TLP and elsewhere, it is clear that is not always working well or provides a wanted benefit for the user. 

I think it would be a better idea to enable TLP discriminatively for select hardware where it is verified to be working well, rather than to enable it across the board.

Mic92, 2025-04-30:
At least if you use `config.services.power-profiles-daemon.enable` it's not used which is the case for gnome/kde etc.

Mic92, 2025-04-30:
But potentially we can invert the default and switch to power-profiles-daemon by default?

jolars, 2025-04-30:
On Wed Apr 30, 2025 at 9:15 AM CEST, Jörg Thalheim wrote:
> Mic92 left a comment (NixOS/nixos-hardware#1465)
>
> At least if you use `config.services.power-profiles-daemon.enable` it's not used which is the case for gnome/kde etc.

Yes, and it's also easy to disable by just using `programs.tlp.enable = false`, but I think it shouldn't be enabled by default.

TLP is pretty aggressive and I don't need it on my Thinkpad t14 gen 5. After starting to use nixos-hardware, it
took me quite a bit of time before I realized that TLP had been enabled and was causing my laptop to
slow down. I'm sure that many users want this kind of power management, but I thought that nixos-hardware
was more of a set of hardware configurations that (more or less) everyone could agree on, and I'm not
really sure TLP is that.

Mic92, 2025-04-30:
I think the choice to use tlp in nixos-hardware is also older than power-profiles-daemon. At the time this might have been the right solution for the hardware back than.

ciferkey, 2025-08-27:
To add another data point, tlp will be enabled by default on the Framework 13, and Framework [explicitly recommends](https://knowledgebase.frame.work/en_us/optimizing-ubuntu-battery-life-Sye_48Lg3) not using tlp.