NixOS/nixos-hardware #1546: Raspberry pi 4 tc358743 device tree overlay fail to build

hatch01, 2025-07-19:
When enabling : 
```nix
hardware.raspberry-pi."4" = {
    tc358743.enable = true;
  };
```

rebuilding return : 
```log
Processing source device tree broadcom/bcm2711-rpi-4-b.dtb...
  Applying overlay tc358743-overlay
Traceback (most recent call last):
  File "/nix/store/3w8w3va9xc0qr91d849v25rh8iivhipv-apply_overlays.py", line 110, in <module>
    main()
  File "/nix/store/3w8w3va9xc0qr91d849v25rh8iivhipv-apply_overlays.py", line 100, in main
    dt = apply_overlay(dt, overlay.fdt)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/3w8w3va9xc0qr91d849v25rh8iivhipv-apply_overlays.py", line 57, in apply_overlay
    raise FdtException(err)
libfdt.FdtException: pylibfdt error -1: FDT_ERR_NOTFOUND
```

And when enabling `apply-overlays-dtmerge.enable`
```log
./broadcom -> /nix/store/iaibhhb2s13hfdvck2gy24yvnx64y52a-device-tree-overlays/./broadcom
'./broadcom/bcm2711-rpi-4-b.dtb' -> '/nix/store/iaibhhb2s13hfdvck2gy24yvnx64y52a-device-tree-overlays/./broadcom/bcm2711-rpi-4-b.dtb'
'./broadcom/bcm2711-rpi-400.dtb' -> '/nix/store/iaibhhb2s13hfdvck2gy24yvnx64y52a-device-tree-overlays/./broadcom/bcm2711-rpi-400.dtb'
'./broadcom/bcm2711-rpi-cm4-io.dtb' -> '/nix/store/iaibhhb2s13hfdvck2gy24yvnx64y52a-device-tree-overlays/./broadcom/bcm2711-rpi-cm4-io.dtb'
Applying overlay tc358743-overlay to bcm2711-rpi-4-b.dtb... DTOVERLAY[error]: can't find symbol 'i2c_csi_dsi'
```

I have cherry-picked in nixpkgs commits, and it fails to build at the commit : 228b2cd449a61ca94abf7b5fd5b27a48987a888d which upgrades from 1.20230106 -> 1.20230405.

I have checked diff of the device tree in raspberry pi between those versions: 
`git diff 1.20230106 1.20230405 -- arch/arm/boot/dts/bcm2711-rpi-4-b.dts`
```diff
diff --git a/arch/arm/boot/dts/bcm2711-rpi-4-b.dts b/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
index 8c0ab39beea1..3afc315077b1 100644
--- a/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
+++ b/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
@@ -1,7 +1,14 @@
 // SPDX-License-Identifier: GPL-2.0
 /dts-v1/;
+#define BCM2711
+#define i2c0 i2c0if
 #include "bcm2711.dtsi"
+#include "bcm283x-rpi-wifi-bt.dtsi"
+#undef i2c0
+#include "bcm270x.dtsi"
+#define i2c0 i2c0mux
 #include "bcm2711-rpi.dtsi"
+#undef i2c0
 //#include "bcm283x-rpi-usb-peripheral.dtsi"
 
 / {
@@ -26,11 +33,6 @@ led-pwr {
 		};
 	};
 
-	wifi_pwrseq: wifi-pwrseq {
-		compatible = "mmc-pwrseq-simple";
-		reset-gpios = <&expgpio 1 GPIO_ACTIVE_LOW>;
-	};
-
 	sd_io_1v8_reg: sd_io_1v8_reg {
 		compatible = "regulator-gpio";
 		regulator-name = "vdd-sd-io";
@@ -56,6 +58,10 @@ sd_vcc_reg: sd_vcc_reg {
 	};
 };
 
+&bt {
+	shutdown-gpios = <&expgpio 0 GPIO_ACTIVE_HIGH>;
+};
+
 &ddc0 {
 	status = "okay";
 };
@@ -65,12 +71,12 @@ &ddc1 {
 };
 
 &expgpio {
-	gpio-line-names = "BT_ON",
+	gpio-line-names = "BT_ON",		/*  0 */
 			  "WL_ON",
 			  "PWR_LED_OFF",
 			  "GLOBAL_RESET",
 			  "VDD_SD_IO_SEL",
-			  "CAM_GPIO",
+			  "CAM_GPIO",		/*  5 */
 			  "SD_PWR_ON",
 			  "SD_OC_N";
 };
@@ -84,66 +90,66 @@ &gpio {
 	 * "FOO" = GPIO line named "FOO" on the schematic
 	 * "FOO_N" = GPIO line named "FOO" on schematic, active low
 	 */
-	gpio-line-names = "ID_SDA",
+	gpio-line-names = "ID_SDA",		/*  0 */
 			  "ID_SCL",
 			  "SDA1",
 			  "SCL1",
 			  "GPIO_GCLK",
-			  "GPIO5",
+			  "GPIO5",		/*  5 */
 			  "GPIO6",
 			  "SPI_CE1_N",
 			  "SPI_CE0_N",
 			  "SPI_MISO",
-			  "SPI_MOSI",
+			  "SPI_MOSI",		/* 10 */
 			  "SPI_SCLK",
 			  "GPIO12",
 			  "GPIO13",
 			  /* Serial port */
 			  "TXD1",
-			  "RXD1",
+			  "RXD1",		/* 15 */
 			  "GPIO16",
 			  "GPIO17",
 			  "GPIO18",
 			  "GPIO19",
-			  "GPIO20",
+			  "GPIO20",		/* 20 */
 			  "GPIO21",
 			  "GPIO22",
 			  "GPIO23",
 			  "GPIO24",
-			  "GPIO25",
+			  "GPIO25",		/* 25 */
 			  "GPIO26",
 			  "GPIO27",
 			  "RGMII_MDIO",
 			  "RGMIO_MDC",
 			  /* Used by BT module */
-			  "CTS0",
+			  "CTS0",		/* 30 */
 			  "RTS0",
 			  "TXD0",
 			  "RXD0",
 			  /* Used by Wifi */
 			  "SD1_CLK",
-			  "SD1_CMD",
+			  "SD1_CMD",		/* 35 */
 			  "SD1_DATA0",
 			  "SD1_DATA1",
 			  "SD1_DATA2",
 			  "SD1_DATA3",
 			  /* Shared with SPI flash */
-			  "PWM0_MISO",
+			  "PWM0_MISO",		/* 40 */
 			  "PWM1_MOSI",
 			  "STATUS_LED_G_CLK",
 			  "SPIFLASH_CE_N",
 			  "SDA0",
-			  "SCL0",
+			  "SCL0",		/* 45 */
 			  "RGMII_RXCLK",
 			  "RGMII_RXCTL",
 			  "RGMII_RXD0",
 			  "RGMII_RXD1",
-			  "RGMII_RXD2",
+			  "RGMII_RXD2",		/* 50 */
 			  "RGMII_RXD3",
 			  "RGMII_TXCLK",
 			  "RGMII_TXCTL",
 			  "RGMII_TXD0",
-			  "RGMII_TXD1",
+			  "RGMII_TXD1",		/* 55 */
 			  "RGMII_TXD2",
 			  "RGMII_TXD3";
 };
@@ -178,23 +184,6 @@ &pwm1 {
 	status = "okay";
 };
 
-/* SDHCI is used to control the SDIO for wireless */
-&sdhci {
-	#address-cells = <1>;
-	#size-cells = <0>;
-	pinctrl-names = "default";
-	pinctrl-0 = <&emmc_gpio34>;
-	bus-width = <4>;
-	non-removable;
-	mmc-pwrseq = <&wifi_pwrseq>;
-	status = "okay";
-
-	brcmf: wifi@1 {
-		reg = <1>;
-		compatible = "brcm,bcm4329-fmac";
-	};
-};
-
 /* EMMC2 is used to drive the SD card */
 &emmc2 {
 	vqmmc-supply = <&sd_io_1v8_reg>;
@@ -237,13 +226,6 @@ &uart0 {
 	pinctrl-names = "default";
 	pinctrl-0 = <&uart0_ctsrts_gpio30 &uart0_gpio32>;
 	uart-has-rtscts;
-	status = "okay";
-
-	bluetooth {
-		compatible = "brcm,bcm43438-bt";
-		max-speed = <2000000>;
-		shutdown-gpios = <&expgpio 0 GPIO_ACTIVE_HIGH>;
-	};
 };
 
 /* uart1 is mapped to the pin header */
@@ -261,12 +243,13 @@ &vec {
 	status = "disabled";
 };
 
+&wifi_pwrseq {
+	reset-gpios = <&expgpio 1 GPIO_ACTIVE_LOW>;
+};
+
 // =============================================
 // Downstream rpi- changes
 
-#define BCM2711
-
-#include "bcm270x.dtsi"
 #include "bcm271x-rpi-bt.dtsi"
 
 / {
@@ -282,7 +265,7 @@ soc {
 
 / {
 	chosen {
-		bootargs = "coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1";
+		bootargs = "coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0";
 	};
 
 	aliases {
@@ -517,18 +500,19 @@ &gpio {
 	audio_pins: audio_pins {
 		brcm,pins = <40 41>;
 		brcm,function = <4>;
+		brcm,pull = <0>;
 	};
 };
 
 &leds {
 	act_led: led-act {
-		label = "led0";
+		default-state = "off";
 		linux,default-trigger = "mmc0";
 		gpios = <&gpio 42 GPIO_ACTIVE_HIGH>;
 	};
 
 	pwr_led: led-pwr {
-		label = "led1";
+		default-state = "off";
 		linux,default-trigger = "default-on";
 		gpios = <&expgpio 2 GPIO_ACTIVE_LOW>;
 	};
@@ -538,7 +522,7 @@ &pwm1 {
 	status = "disabled";
 };
 
-&audio {
+&vchiq {
 	pinctrl-names = "default";
 	pinctrl-0 = <&audio_pins>;
 };
@@ -552,6 +536,8 @@ cam0_reg: &cam_dummy_reg {
 
 / {
 	__overrides__ {
+		audio = <&chosen>,"bootargs{on='snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1',off='snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=0'}";
+
 		act_led_gpio = <&act_led>,"gpios:4";
 		act_led_activelow = <&act_led>,"gpios:8";
 		act_led_trigger = <&act_led>,"linux,default-trigger";
```

There is nothing in this diff that could have broken i2c_csi_dsi.

However, with some die and retry I found that replacing : 
```devicetree
            __fixups__ {
              i2c_csi_dsi = "/fragment@0:target:0";
              csi1 = "/fragment@1:target:0";
              i2c0if = "/fragment@4:target:0";
              i2c0mux = "/fragment@5:target:0";
            };
```
with : 
```devicetree
            __fixups__ {
              i2c1 = "/fragment@0:target:0";
              csi1 = "/fragment@1:target:0";
              i2c0 = "/fragment@4:target:0";
              i2c0mux = "/fragment@5:target:0";
            };
```

I get a successful compilation but not a working device at all.

Also, no diff between the two versions in the files (in the Raspberry Pi kernel): 
- arch/arm/boot/dts/overlays/tc358743-audio-overlay.dts
- arch/arm/boot/dts/overlays/tc358743-overlay.dts
- arch/arm/boot/dts/overlays/tc358743-pi5-overlay.dts
- arch/arm/boot/dts/overlays/tc358743.dtsi

It is my first time dealing with devicetree, so I may not see something obvious.
Any help would be very welcome !

hatch01, 2025-07-19:
@tfc How did you create this overlay (4 years ago yeah)

tfc, 2025-07-20:
Hey @hatch01 i don't remember all details, but this other PR describes how i got DTB related stuff to work:

https://github.com/NixOS/nixos-hardware/pull/264

tfc, 2025-07-20:
There is however an i2c option, too. did you enable that, or only the tc thingy? 

(If that fixes it, please upstream an assertion in the tc module that helps the next person who runs into that problem)

hatch01, 2025-07-20:
https://github.com/NixOS/nixos-hardware/commit/587339210cecce66cd2c81d71c1965f26fb44ad4
Seems to be ok.
I'll look if I can apply it to every RPi4 overlay